AutoRally/autorally #2: ROS simulation not starting

p9x, 2016-05-20:
Hi, i am new to ROS but i did the setup accordingly on a vmware 1404 client.

on the last roslaunch simulation, it was not successful:

... logging to /home/desk/.ros/log/a2cc5d5a-1eaf-11e6-9179-000c29410603/roslaunch-ubuntu-41202.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

error loading <rosparam> tag: 
	file does not exist [/config//servoCommandPriorities.yaml]
XML is <rosparam command="load" file="$(env AR_CONFIG_PATH)//servoCommandPriorities.yaml" param="servoCommandProirities"/>
The traceback for the exception was written to the log file

could you please help me with this, thanks.

barulicm, 2016-05-20:
Try replacing line _94_ of `autorally/autroally_gazebo/launch/autoRallyTrackGazeboSim.launch` with the following:

`<rosparam param="servoCommandProirities" command="load" file="$(env AR_CONFIG_PATH)/servoCommandPriorities.yaml" />`

The release version mistakenly has two forward slash characters before the file name.

If this does not resolve the issue, please confirm that you still have the `servoCommandPriorities.yaml` file in `autorally/autorally_util/config/`.

p9x, 2016-05-21:
Thanks for your advice, after
`source autorally/autorally_util/setupEnvLocal.sh`
Gazebo successfully initialized.

However the auto model loading has some problem so i have it downloaded manually. (gazebo v2.2.3)

Now i will need to build an arduino joystick to test the simulation, thanks.

barulicm, 2016-05-25:
Glad to hear you were able to get Gazebo started. Can you be any more specific about the problems your were having with model loading?

closed 2016-08-03 by barulicm