AutoRally/autorally #3: Platform build instructions missing?

bikegriffith, 2016-05-23:
This project looks amazing! I just noticed that your web site build instructions (https://autorally.github.io/build/) has a broken link for the platform build instructions (supposed to be https://github.com/AutoRally/autorally_platform_instructions but perhaps this repo is not public).  

Do you plan to publish the platform instructions? 

Thank you!

bgoldfai, 2016-05-23:
We're a little behind getting the instructions repository organized, so it's not yet pubic. We'll make it public with the first few documents (chassis instructions, parts list, some of the CAD models) this week, and add the rest of the documents as we complete them.

Keep in mind that since the platform is still under active development, we will tweak the instructions from time to time.

bgoldfai, 2016-05-25:
The initial set of instructions are now public. Keep in mind these are not yet complete, and will be updated throughout the summer.

closed 2016-05-25 by bgoldfai