AutoRally/autorally #4: Remove unnecessary null pointer checks

elfring, 2016-05-23:
[An extra null pointer check is not needed](http://dietmar-kuehl.de/mirror/c++-faq/freestore-mgmt.html#faq-16.8 "Do I need to check for NULL before delete p?") in [the functions "QNode::imageCallback1" and "QNode::imageCallback2"](https://github.com/AutoRally/autorally/blob/4508587abaf8dc7956dc6356773b84cc0f697754/autorally_core/src/ocs/qnode.cpp#L248 "Update candidates").

barulicm, 2016-05-24:
Thanks for pointing this out. I've removed those checks in our development branch. This issue will automatically close with our next release.

closed 2016-07-28 by bgoldfai

elfring, 2016-07-29:
Thanks for [your small source code improvement](https://github.com/AutoRally/autorally/commit/577ee695c96093effa2348094862a1d8d6e98313 "Deletion of two unnecessary checks").

commit 577ee695 (2016-05-24): Removes unnecessary null checks in qnode.
Fixes #4
commit db816522 (2016-08-01): v0.2.0 (#28)

* Moving deprecated Eigen to Eigen3 find package

* Templatizes setupEnvVariables.sh

* Adds support for launching cameras with only one camera connected.

* Fixes config path typo in simulator launch files.

* Removes unnecessary null checks in qnode.
Fixes #4

* Refactors OCS image callbacks, pulling image formatting into a separate function.
Fixes #5

* Adds error handling with console output to pthread mutex init and lock calls.
Fixes #6

* added panos and evangelos websites

* Added yaw rate to published state?

* Fixes state estimator topic remapping.

* Correct gyro with e [... cut]
commit 545ea8f8 (2016-08-04): v0.2.2 (#32)

* Moving deprecated Eigen to Eigen3 find package

* Templatizes setupEnvVariables.sh

* Adds support for launching cameras with only one camera connected.

* Fixes config path typo in simulator launch files.

* Removes unnecessary null checks in qnode.
Fixes #4

* Refactors OCS image callbacks, pulling image formatting into a separate function.
Fixes #5

* Adds error handling with console output to pthread mutex init and lock calls.
Fixes #6

* added panos and evangelos websites

* Added yaw rate to published state?

* Fixes state estimator topic remapping.

* Correct gyro with e [... cut]