AutoRally/autorally #7: MPPI controller information

yashim, 2016-05-28:
Hi!
Do you plan to share the MPPI controller code (mppi_controller)? Where can I find the information/theory about it?

bgoldfai, 2016-05-28:
We plan to release the MPPI controller as a standalone repository so people can use the controller with whatever system they choose (including the AutoRally platform).

The theory behind the controller is detailed in the publication "Aggressive Driving with Model Predictive Path Integral Control" presented two weeks ago at ICRA 2016. The paper should be available in the next few weeks.

yashim, 2016-06-02:
Thank you for your answer!
Could you please give an estimate how much time will it take to release MPPI controller?

bgoldfai, 2016-06-03:
Hopefully by the end of the summer, but right now we're not sure.

bgoldfai, 2017-06-12:
The MPPI code is now publicly available in the devel branch, and will be included in the next release.

closed 2017-06-12 by bgoldfai