AutoRally/autorally #10: gazebo errors

LiliMeng, 2016-06-01:
Hi, thanks for the open-source code!

It has the following errors after $roslaunch autorally_gazebo autoRallyTrackGazeboSim.launch

Error [Server.cc:345] Unregistered physics engine [dart], the default will be used instead.

Msg Connected to gazebo master @ http://127.0.0.1:11345
Msg Publicized address: 206.87.212.231
Error [Connection.hh:264] Header is empty
Error [Master.cc:124] Master got empty data message from[45889]
Error [Connection.hh:264] Header is empty
Error [Master.cc:124] Master got empty data message from[45889]
Error [Connection.hh:264] Header is empty
Error [Master.cc:124] Master got empty data message from[45889]
Error [Connection.hh:264] Header is empty
Error [Master.cc:124] Master got empty data message from[45889]
Error [Connection.hh:264] Header is empty
Error [Master.cc:124] Master got empty data message from[45889]
Error [Connection.hh:264] Header is empty
Error [Master.cc:124] Master got empty data message from[45889]
Error [Connection.hh:264] Header is empty
Error [Master.cc:124] Master got empty data message from[45889]
Error [Connection.hh:264] Header is empty
Error [Master.cc:124] Master got empty data message from[45889]

bgoldfai, 2016-06-02:
Do these errors cause the simulation to not work?

- The unregistered physics engine shouldn't be a problem since Gazebo switches to a default one if you don't have DART compiled into Gazebo, and we don't use any DART-specific features.
- I sometimes see the other repeated errors when I run the simulation, but haven't taken the time to track them down because they don't appear to affect the simulation. They seem to be a results of some sort of startup ordering that takes care of itself once everything is up and running.

bgoldfai, 2016-07-28:
This seems to be stale, I'll reopen if the issue arises again.

closed 2016-07-28 by bgoldfai