AutoRally/autorally #13: Running with Kinetic on Ubuntu 16.04

bgoldfai, 2016-06-09:
Open issues in 16.04 with kinetic:
packages that must install from source:
- camera1394
- driver_common
- hector_gazebo (a gazebo 7 compatible branch)
- ros_control (not completely sure if this is required, but can't get simulation to start with ros-kinetic-controller-manager through cmd line)

before full compilation, compile driver_base with: `catkin_make --pkg driver_base`

our own nodes that don't compile right now:
- stateEstimator

Joint controllers for simulation don't seem to work, needs more debugging

ShawnSchaerer, 2016-07-27:
@bgoldfai any progress on the 16.04 kinetic build?

bgoldfai, 2016-07-29:
I can drive the platform around in simulation with a gamepad using our joystick node if I have the following packages installed from source, still explicitely compiling driver_base first:
* driver_common
* gazebo_ros_pkgs

Camera1394 is no longer a dependency, but hasn't been removed from package.xml yet. 

The state estimator still does not compile because of GTSAM dependency issues in 16.04.

We have not tested 16.04/kinetic on hardware, as all of our compute boxes are still on 14.04/indigo. We will migrate once all of these remaining issues are resolved.

bgoldfai, 2016-07-29:
After further testing, there is no need for any ROS packages to be installed from source and the state estimator compiles in 16.04 if GTSAM is compiled from the latest develop branch.

The standard installation instructions for our software are now compatible with 16.04/kinetic in the kinetic-devel branch. We won't officially move up to kinetic until we have a chance to test on hardware.

ShawnSchaerer, 2016-08-01:
Great.  I'll try it out

ShawnSchaerer, 2016-08-08:
I have a compile issue with 16.04 and kinetic.  any ideas??

Linking CXX executable /home/northstar/catkin_ws/devel/lib/autorally_core/imuGpsEstimator
/usr/bin/ld: CMakeFiles/imuGpsEstimator.dir/IMU_GPS.cpp.o: undefined reference to symbol '_ZN3tbb8internal23allocate_via_handler_v3Em'
//usr/lib/x86_64-linux-gnu/libtbb.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status

bgoldfai, 2016-08-08:
I was able to replicate the issue and I think solve it by linking TBB required by GTSAM during the state estimator compilation. The kinetic-devel branch is updated with this change.

ShawnSchaerer, 2016-08-08:
just tested it and it compiled.  I will start to test the running system.

sammilei, 2017-12-12:
put the driver_base in the workspace/src and compile. It worked for me.
driver_base source is here: https://github.com/ros-drivers/driver_common

bgoldfai, 2018-03-13:
The repository has been tested working on quite a few machines using the instructions in the kinetic-devel branch, which is now the default, so I'm closing this issue.

closed 2018-03-13 by bgoldfai