AutoRally/autorally #14: IMU orientation transform

pdrews, 2016-06-13:
Verify/fix the transform from the IMU internal navigation frame (published on /imu/filter) to the state estimator navigation frame.  This issue may be caused by a poorly calibrated magnetometer inside the IMU, or a problem with the transform.  More investigation is needed to determine.

myxyzy, 2021-02-09:
Hello, 
I was looking for car sensors and come across with this IMU  issue. Is this still open issue?

JasonGibson274, 2021-02-09:
I'll let @pdrews answer to be fully sure but I do not think so. Our localization stack works as expected so I think it was a transform issue.

pdrews, 2021-02-09:
I'm not sure what the original intent of this issue was, but it currently works fine so we can close it.

closed 2021-02-09 by pdrews