AutoRally/autorally #31: Missing chassisCommandPriorities config

sfowlr, 2016-08-04:
Autorally Gazebo launch file autoRallyTrackGazeboSim.launch calls for chassisCommandPriorities.yaml on line 89:
`<rosparam param="chassisCommandProirities" command="load" file="$(env AR_CONFIG_PATH)/chassisCommandPriorities.yaml" />`

However the file doesn't exist. The following error appears in the terminal:
`error loading <rosparam> tag: 
	file does not exist [/home/zzz/catkin_ws/src/autorally/autorally_util/config/chassisCommandPriorities.yaml]
XML is <rosparam command="load" file="$(env AR_CONFIG_PATH)/chassisCommandPriorities.yaml" param="chassisCommandProirities"/>
The traceback for the exception was written to the log file`

bgoldfai, 2016-08-04:
This is an issue that we fixed, and then reintroduced by accident. 

As a temporary fix you can change `chassisCommandPriorities.yaml` to `servoCommandPriorities.yaml1

I will push an updated version (0.2.2) to GitHub in the next hour after I test it.

sfowlr, 2016-08-04:
I read through the previous issue where it was mentioned and suspected that may be the cause. Thanks for the quick response.

commit 1d5943a4 (2016-08-04): Fixes #31, which was a reintroduction of issue #2 by
changing the command priorities file from servoCommandPriorities.yaml to chassisCommandPriorities.yaml
closed 2016-08-04 by bgoldfai