AutoRally/autorally #34: Simulation Instructions / Tutorial

ShawnSchaerer, 2016-08-10:
I think there is a need to have instructions on how to run the system in Gazebo.
I can run the simulator after installing as per the document, but its not clear on how to launch the rest of the system to run in simulation mode.

I tried to launch the autorally_platform but there are too many dependencies on physical hardware and the simulation does not work.

Is there a simulation tutorial somewhere?
Thanks

bgoldfai, 2016-08-10:
Yeah we have some tutorials on the wiki to run the system autonomously that work in simulation and with a physical platform. If you go through them in the following order (one tutorial for each node involved) you should have the vehicle driving autonomously in simulation:
- State Estimator
- Waypoint Follower
- Constant speed Controller

Would something like an overall tutorial or better section at the end of the README that lays that procedure out explicitly help? Or is there something larger I'm missing that we should add tutorial wise?

ShawnSchaerer, 2016-08-10:
Thanks,
I will go through and let you know what I think.  It was not clear to me on how to proceed base on the ##What's Next  on the bottom of the readme.

Maybe just a better section at the end of the readme for now.

bgoldfai, 2016-08-10:
OK I'll add to that section.

ShawnSchaerer, 2016-08-10:
I tried to run those tutorials in order but I could not get the car to move in gazebo  (ROS Kinetic on Ubuntu 16.04).  I did get it to move when I set the speed manually using rqt_graph.
Perhaps I have not configured everything correctly or there is a bug with Kinetic?
I will try again tomorrow.

bgoldfai, 2016-08-10:
On my 16.04 test setup it looks like the simulated GPS and IMU aren't publishing data looking at topics /gpsRoverStatus and /imu/imu (I don't even see these topics). Without those the state estimator and the waypoint follower won't function. I'll continue to debug, this may be an issue with hector_gazebo_plugins or a new way to use the plugins in Kinetic.

The constant speed controller appears to control the throttle, as it only relies on the /wheelSpeeds topic. If you are using a gamepad, make sure motionEnabled in the /runstop message coming from the joystick node is true. You can toggle it with any of the buttons next to the right stick.

If you aren't using a gamepad:
- comment out line 93 of autorally_gazebo/launch/autoRallyTrackGazeboSim.launch
- launch the ocs `roslaunch autorally_core ocs.launch` - this is because the chassis controller requires a runstop message as a heartbeat, and the OCS publishes one
- press the red button labelled **Runstop Motion Disabled** to enable motion through the OCS runstop message
- verify runstop motion is enabled by looking the runstop message (watching the /runstop and looking at the **Runstop Motion En** field in the OCS, all of which should be true)

ShawnSchaerer, 2016-08-10:
Thanks,  I will try the constant speed controller and see what happens.

bgoldfai, 2016-08-10:
OK so I think I solved the issue by publishing the simulated IMU and GPS on the expected topics directly instead of remapping them. The most concise procedure to get the platform driving autonomously once the simulator is up and running with the robot positioned where it starts, and you have verified that /runstop enableMotion is true from your /runstop source:

Start state estimator:
- in `autorally_core/launch/state_estimator.launch` change `InvertY` and `InvertZ` to **false**
- `rosparam set /gps_imu/FixedInitialPose true`
- `roslaunch autorally_core state_estimator.launch`

Start waypoint follower:
- `roslaunch autorally_control waypointFollower.launch`

Start constant speed controller:
- `roslaunch autorally_control constantSpeedController.launch`
- `rosrun rqt_publisher rqt_publisher`
-- publish topic `constantSpeedController/speedCommand` of type `std_msgs_Float64` at rate 10 with value of 3. This is the target velocity in m/s and as soon as you do this the platform should move, assuming motion is enabled.

If the robot turns and hits the barrier it's probably because the state estimator is not converged, so its orientation estimate is incorrect. Just select the track barriers and move them up to allow the robot to continue driving, and the estimator should converge and the vehicle will return to within the barriers.

ShawnSchaerer, 2016-08-10:
Ok thanks.  I will try it out and post results

ShawnSchaerer, 2016-08-11:
I verified that this procedure works.  
Its good enough for me to learn the code base now. :)

I would suggest that we put that procedure for the sim in the readme.

bgoldfai, 2016-08-11:
Awesome, thanks for taking the time to work through all this with me. I'll add that procedure to the readme then refer to the wiki for more information about each of the nodes for the curious.

ShawnSchaerer, 2016-08-11:
no problem.
I am going to be adapting parts of the system to a new platform and trying
to see what happens if we swap out the GTSAM with Robot_localization.

great work so far :)

dstoyer, 2017-01-24:
This is awesome work, so my following comment is intended to hopefully improve the installation experience.

I have installed the autorally software platform in Ubuntu 14.04 and Unbuntu 16.04 (using the kinetic-devel branch). In both cases I had to clone imu_3dm_gx4 into catkin_ws/src. 
Installation went smoothly by following the current instructions with these steps included:
_After_ cloning the appropriate autorally into catkin_ws/src
1. clone imu_3dm_gx4 (or download and unzip) into the same directory
2. run "rosdep install --from-path src --ignore-src -y" from catkin_ws
_Continue_ by running catkin_make from catkin_ws/

I don't know if imu_3dm_gx4 is _supposed_ to be installed as a dependency automatically with the "rosdep install ..." line, but it currently is not. 

I have tested the install an 3 different systems and VMware workstation. The install is successful on all of them when adding those two steps (for Gazebo in VMware, OpenGL needs to be set to 2.0: export SVGA_VGPU10=0).

Again, you guys are doing awesome work, hopefully this will either clarify the installation instructions, or point out a bug in the installation implementation.

Cheers,
Dan

bgoldfai, 2017-01-25:
Hi, thanks for the comment and the enthusiasm for the project!

To make sure I understand correctly, you are working with the simulated AutoRally platform in Gazebo and not on a physical AutoRally platform, and the installation/compilation process fails if the imu_3dm_gx4 repository is not cloned into your catkin workspace using the 2 steps you describe?

I ask because the instructions used to include cloning imu_3dm_gx4 exactly where you are suggesting, but that step was moved into the first section of the [platform configuration instructions](https://github.com/AutoRally/autorally/wiki/Platform%20Configuration%20Instructions) because that repository is (in theory) only needed if you are installing on a physical AutoRally platform with a 3DM-GX4-25 IMU. The simulated IMU comes from [hector_gazebo_plugins](http://wiki.ros.org/hector_gazebo_plugins) which should automatically install in the `rosdep install` step.

dstoyer, 2017-01-25:
> To make sure I understand correctly, you are working with the simulated AutoRally platform in Gazebo and not on a physical AutoRally platform, and the installation/compilation process fails if the imu_3dm_gx4 repository is not cloned into your catkin workspace using the 2 steps you describe?

Correct.

> the instructions used to include cloning imu_3dm_gx4 exactly where you are suggesting, but that step was moved into the first section of the platform configuration instructions

Admittedly, I did not go through the platform installation instructions. I did skim over it but moved on to the software instructions since I don't have any of the hardware and saw that the software could be run in simulation.

> that repository is (in theory) only needed if you are installing on a physical AutoRally platform with a 3DM-GX4-25 IMU.

Specifics on issues I ran into without imu_3dm_gx4 present:

* During installation <code>autorally/autorally_core/src/StateEstimator/CMakeLists.txt</code> checks for GTSAM at line 7. When using <code>catkin_make</code> from the autorally catkin workspace, if GTSAM is  installed there are compilation errors stating that <code>imu_3dm_gx4/FilterOutPut.h</code> cannot be found. This is generated by line 68 of <code>autorally/autorally_core/src/StateEstimator/IMU_GPS.h</code> which is compiled because the above CMakeLists.txt checks for GTSAM and finds it (and obviously IMU_GPS.cpp is heavily reliant on imu_3dm_gx4).

* If GTSAM is not installed, <code>catkin_make</code> compiles autorally without errors, however, when <code>roslaunch autorally_core stateEstimator.launch</code> is run, the error <code>cannot launch node of type [autorally_core/imuGpsEstimator]: can't
locate node [imuGpsEstimator] in package [autorally_core]</code> appears.

* Further, I'm not sure if GTSAM is required to run the simulation. If it is, then the installation currently requires the imu_3dm_gx4 package. If GTSAM is not required, then the installation proceeds normally, but it appears <code>imuGpsEstimator</code> is not being created with hector_gazebo_plugins.

* I was not looking for hector_gazebo_plugins during <code>rosdep install</code>, so I can't say if that installed properly or not. It will probably be a few days before I can get back to that.

Hope that helps,

Dan

bgoldfai, 2017-01-27:
This is all great and thanks for taking the time to put this information together!

I added your suggested step back into the instructions since the state estimator does need a custom message from the imu repo looking at the error message you provided and the code (I didn't think we had that dependency and didn't check compilation on a machine without the imu code). That change is in the devel branch now, and should make its way into the kinetic-devel next week.

Strictly speaking, GTSAM is not required for the simulation world to run. GTSAM is required to compile and run the state estimator, which provides a fused state estimate for the robot and is required to run the waypoint follower.

Let us know if you have any other questions or issues.

ksoltani, 2017-12-12:
Hi all,
Thanks for the discussions. I have been trying to get the car execute wayPointFollower. I set the speed using the constantSpeedController and through rqt_publisher. The car does move but it never steers. I suspect somehow wayPointFollower is not actually changing the state of the car. Have you tried this? Did you have similar issues?

bgoldfai, 2017-12-20:
What OS, ROS version, and autorally branch are you working in, and are any errors generated on the command line that you can provide for debugging? I think there is a known issue in the kinetic_devel branch where the waypoint follower crashes on launch that I'll look into if I can replicate it on my machine.

Another thing to check is to look at the steeringCommander in the /chassisStatus topic to see what is in control of the steering to verify that commands from the waypoint follower are being received.

ksoltani, 2017-12-21:
Hi Brian,
I was using grady-kinetic-devel branch. I am running Ubuntu 16.04, ROS Kinetic.
Once I used kinetic-devel branch I was able to get it to execute wayPointFollower control using throttle setting in ocs. I did not test with speed controller.

I also remember I had to change OpenCV version requirement to 3 instead of 2 before it could build the code.

bgoldfai, 2017-12-27:
I'm glad kinetic-devel worked, thanks for letting us know. We've encountered the OpenCV version problem as well and plan to fix it in the next release and remove the OpenCV exact version requirement in the CMakeLists file.

bgoldfai, 2018-03-13:
The suggestions in this issue have been incorporated into the current kinetic-devel branch, everything seems to be working, so I'm closing.

closed 2018-03-13 by bgoldfai