AutoRally/autorally #38: Where can I find the parameters of car's dynamic?

yunxiaoshan, 2016-09-27:
Hi, 
I am so exited I can run the platform, this is the first plantform I can run since I start to learn ros. I am trying to implant a pure pursuit tracking controller, it shuold be very simple. But I need some car's parameters. And also, I want to further extend the motion planning to this platform, while it should be a big challenge for me because of my weak knowledge of ros, I will keep trying.
-shan

bgoldfai, 2016-10-04:
Hi Shan, that's great you were able to get it running! Many of the vehicle parameters can be found in the [robot description file](https://github.com/AutoRally/autorally/blob/master/autorally_description/urdf/autoRallyPlatform.urdf.xacro), but that file is pretty messy. I can add a table of vehicle parameters if you give me the specific parameters you need.

closed 2021-02-04 by JasonGibson274

KarasevPavel, 2023-06-23:
Hello @bgoldfai!

Could You help me with car model parameters and tire parameters (Pacejka tire coefficients) of the car, which were estimated in this [**article**](https://dcsl.gatech.edu/papers/acc17b.pdf)?

Parameters, that I am looking for are:
1) Pacejka tire coefficients (B, C, D, E, Sh, Sv, Fz_nominal);
2) Car model parameters (inertia Iz)