AutoRally/autorally #54: Compiling error with GPU

qazmichaelgw, 2017-11-25:
I use the kinetic-devel branch in https://github.com/AutoRally/autorally.git. I compile it in Ubuntu 16.04. There is some compiling error with respect to CUDA in the autorally_control package. 

I got the following error:
```/home/gaowei/catkin_ws/src/autorally/autorally_control/include/autorally_control/path_integral/generalized_linear.cut(98): error: a pointer to a bound function may only be used to call the function

/home/gaowei/catkin_ws/src/autorally/autorally_control/include/autorally_control/path_integral/neural_net_model.cut(79): error: a pointer to a bound function may only be used to call the function

/home/gaowei/catkin_ws/src/autorally/autorally_control/include/autorally_control/path_integral/neural_net_model.cut(80): error: a pointer to a bound function may only be used to call the function

/home/gaowei/catkin_ws/src/autorally/autorally_control/include/autorally_control/path_integral/generalized_linear.cut(98): error: a pointer to a bound function may only be used to call the function
```

gradyrw, 2017-11-27:
This is actually just a normal c++ error, not a CUDA error. It looks like the cnpy api (https://github.com/rogersce/cnpy), which is used to load python numpy arrays, changed recently. You're likely using the new version where "data" is now a member function instead of a data member. This is what's causing the error.

If you change:
 double* foo = reinterpret_cast<double*>(foo_raw.data); 

to instead be:
double* foo = foo_raw.data<double>();

in generalized_linear and neural_net_model then it should fix the problem. We'll need to figure out a way of preventing this type of problem in the future.

qazmichaelgw, 2017-12-02:
Thanks a lot. I got it work by changing it to:
double* theta_ptr = theta_raw.data<double>();

JasonGibson274, 2020-06-12:
Closing issue since no longer relevant with the write that happened in between then and now

closed 2020-06-12 by JasonGibson274