AutoRally/autorally #55: A script to generate track_data.txt file

alexbuyval, 2017-11-27:
I wonder there is any script to generate [track_data.txt](https://github.com/AutoRally/autorally/blob/kinetic-devel/autorally_control/src/path_integral/params/maps/gazebo/track_data.txt) file for new track?

gradyrw, 2017-11-27:
Right now there is not a general purpose script to help with creating a costmap for a new track. If you create a costmap yourself, you can save it in the track_data format and use it with the MPPI controller. The track_data.txt format is the following:

minimum x-value (in meters)
maximum x-value (meters)
minimum y-value (meters)
maximum y-value (meters)
grid resolution (i.e. how many grid points make up 1 meter)
grid values in row-major order

alexbuyval, 2017-11-28:
Thank you for quick response!

closed 2017-11-28 by alexbuyval