AutoRally/autorally #60: [kinetic-devel] Segmentation Fault When Launching Gazebo Simulator

kphawkins, 2018-02-08:
I'm trying to run the Gazebo simulator after a fresh install and getting a crash every time:

`[ WARN] [1518055560.114620641, 0.034000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'EffortJointInterface' within the <hardwareInterface> tag in joint 'right_rear_axle'.
[ INFO] [1518055560.177734688, 0.034000000]: Loaded gazebo_ros_control.
[autorally_platform/spawn_platform-4] process has finished cleanly
log file: /home/kelsey/.ros/log/8d986cc0-0c61-11e8-a754-309c230ef122/autorally_platform-spawn_platform-4*.log
[WARN] [1518055560.347148, 0.034000]: The specified list of shock absorbers is invalid. No shock absorbers will be used.
[WARN] [1518055560.349486, 0.034000]: The specified namespace value is invalid. The default timeout value will be used instead.
Segmentation fault (core dumped)
[autorally_platform/gazebo-2] process has died [pid 20555, exit code 139, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e dart urdf/populated_AR.world /autorally_platform/camera/left:=/left_camera /autorally_platform/camera/right:=/right_camera __name:=gazebo __log:=/home/kelsey/.ros/log/8d986cc0-0c61-11e8-a754-309c230ef122/autorally_platform-gazebo-2.log].
log file: /home/kelsey/.ros/log/8d986cc0-0c61-11e8-a754-309c230ef122/autorally_platform-gazebo-2*.log`

Gazebo works fine when launching independently so I don't believe it's a graphics card/driver issue.  I believe it happens when starting the gazebo server and might be related to an invalid URDF model.

Ubuntu 16.04
ROS Kinetic
Gazebo 7.9.0
Branch: kinetic-devel

kphawkins, 2018-02-08:
It does appear that the autoRallyTrackMultipleCarsGazeboSim.launch works properly.  I can reverse engineer a single car version in the coming days.

karanchawla, 2018-02-08:
Try `killall gzserver && killall gzclient` before launching the simulation.

JasonGibson274, 2018-02-09:
I was able to get this working on two different machines using the code in gazebo_fixes. Let me know if you are still having issues launching on that branch.

kphawkins, 2018-02-09:
That works just fine. Thanks!

closed 2018-02-09 by kphawkins