AutoRally/autorally #63: catkin_make出错 Catkin_make error with Ubuntu 16.04.4

ytgcljj, 2018-03-09:
I use Ubuntu 16.04.4 cuda-8.0

usr/include/boost/type_traits/detail/has_binary_operator.hp:50: Parse error at "BOOST_JOIN"
autorally/autorally_core/src/ocs/CMakeFiles/ocs.dir/build.make:88: recipe for target 'autorally/autorally_core/src/ocs/moc_qnode.cxx' failed
make[2]: *** [autorally/autorally_core/src/ocs/moc_qnode.cxx] Error 1
CMakeFiles/Makefile2:5208: recipe for target 'autorally/autorally_core/src/ocs/CMakeFiles/ocs.dir/all' failed
make[1]: *** [autorally/autorally_core/src/ocs/CMakeFiles/ocs.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
[ 59%] Building CXX object autorally/autorally_core/src/SerialSensorInterface/CMakeFiles/SerialSensorInterface.dir/SerialInterfaceThreaded.cpp.o
[ 59%] Built target driver_base_generate_messages_eus
[ 59%] Building CXX object autorally/autorally_core/src/SerialSensorInterface/CMakeFiles/SerialSensorInterface.dir/SerialCommon.cpp.o
[ 59%] Built target autorally_msgs_generate_messages_eus
[ 60%] Building CXX object imu_3dm_gx4/CMakeFiles/imu_3dm_gx4.dir/src/imu.cpp.o
[ 61%] Linking CXX shared library /home/launchx-mask/catkin_ws/devel/lib/libRingBuffer.so
[ 61%] Built target RingBuffer
[ 62%] Linking CXX shared library /home/launchx-mask/catkin_ws/devel/lib/libSerialSensorInterface.so
[ 62%] Built target SerialSensorInterface
[ 63%] Linking CXX shared library /home/launchx-mask/catkin_ws/devel/lib/libImageRepublisher.so
[ 63%] Linking CXX shared library /home/launchx-mask/catkin_ws/devel/lib/libDiagnostics.so
[ 63%] Built target Diagnostics
[ 63%] Built target ImageRepublisher
[ 63%] Linking CXX executable /home/launchx-mask/catkin_ws/devel/lib/imu_3dm_gx4/imu_3dm_gx4
[ 63%] Built target imu_3dm_gx4
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

bgoldfai, 2018-03-09:
Hi, I have not been able to reproduce that compilation error on our machines with Ubuntu 16.04 and cuda-8.0.

Are you using the kinetic-devel branch?

ytgcljj, 2018-03-09:
I have two questions. One is usr/include/boost/type_traits/detail/has_binary_operator.hp. This path is not a root directory. The other is kinetic-devel. How to execute it is not clear, thank you.

closed 2018-03-12 by ytgcljj