AutoRally/autorally #67: cuda related compile error

yfzhang, 2018-06-30:
Basically followed the installations instructions, but got the following error. Some variables in the .cu files seems not to be defined.

``[ 96%] Building NVCC (Device) object autorally/autorally_control/src/path_integral/CMakeFiles/path_integral_bf.dir/path_integral_bf_generated_path_integral_main.cu.o
/home/yf/autorally_ws/src/autorally/autorally_control/src/path_integral/path_integral_main.cu(81): error: identifier "DynamicsModel" is undefined

/home/yf/autorally_ws/src/autorally/autorally_control/src/path_integral/path_integral_main.cu(81): error: identifier "MPPI_NUM_ROLLOUTS__" is undefined

/home/yf/autorally_ws/src/autorally/autorally_control/src/path_integral/path_integral_main.cu(81): error: identifier "BLOCKSIZE_X" is undefined

/home/yf/autorally_ws/src/autorally/autorally_control/src/path_integral/path_integral_main.cu(81): error: identifier "BLOCKSIZE_Y" is undefined

/home/yf/autorally_ws/src/autorally/autorally_control/src/path_integral/path_integral_main.cu(97): error: identifier "DynamicsModel" is undefined

/home/yf/autorally_ws/src/autorally/autorally_control/src/path_integral/path_integral_main.cu(97): error: identifier "model" is undefined

/home/yf/autorally_ws/src/autorally/autorally_control/src/path_integral/path_integral_main.cu(97): error: expected a type specifier

/home/yf/autorally_ws/src/autorally/autorally_control/include/autorally_control/path_integral/mppi_controller.cuh(56): error: name followed by "::" must be a class or namespace name
          detected during instantiation of class "autorally_control::MPPIController<DYNAMICS_T, COSTS_T, ROLLOUTS, BDIM_X, BDIM_Y> [with DYNAMICS_T=<error-type>, COSTS_T=autorally_control::MPPICosts, ROLLOUTS=<error-constant>, BDIM_X=<error-constant>, BDIM_Y=<error-constant>]" 
/home/yf/autorally_ws/src/autorally/autorally_control/src/path_integral/path_integral_main.cu(103): here
``

nolanwagener, 2018-08-04:
Try using an older version of cmake, like 3.5 or 3.6.

zobertke, 2019-03-18:
It did not work with 3.6 :cry:

farahaniamir08, 2019-04-23:
Hi guys 
I have a issue during compile "catkin_make" .
`/usr/lib/gcc/x86_64-linux-gnu/5/include/mwaitxintrin.h(36): error: identifier "__builtin_ia32_monitorx" is undefined`
I saw same problem in tensor flow GitHub page.[https://github.com/tensorflow/tensorflow/issues/1066](url). I have added this flags to "CMakeLists.txt" exactly like this:
 `set( CMAKE_CXX_FLAGS " -D_MWAITXINTRIN_H_INCLUDED -D_FORCE_INLINES ")` in top of "CMakeLists.txt" after `set(_cmd "catkin_find_pkg" "catkin" "${CMAKE_SOURCE_DIR}")` but doesn't effect.  

I verified my cuda before compile :
       
`+-----------------------------------------------------------------------------+
| NVIDIA-SMI 410.79       Driver Version: 410.79       CUDA Version: 10.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce MX150       Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   53C    P3    N/A /  N/A |    546MiB /  4042MiB |      4%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1149      G   /usr/lib/xorg/Xorg                            28MiB |
|    0      1544      G   /usr/lib/xorg/Xorg                           291MiB |
|    0      1981      G   compiz                                        78MiB |
|    0      2249      G   /proc/self/exe                                18MiB |
|    0      2411      G   ...quest-channel-token=5297298698446075686   103MiB |
+-----------------------------------------------------------------------------+

and 

`nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2015 NVIDIA Corporation
Built on Tue_Aug_11_14:27:32_CDT_2015
Cuda compilation tools, release 7.5, V7.5.17`

Please help me 
Thanks and best regards 
Amir

Diandry, 2019-08-08:
@farahaniamir08 I had the same error message. Make sure your versions of GCC and NVCC are compatible. In my case, using GCC 5.4 Cuda V7.5 is not supported and updating to NVCC 10.1 solved the issue. You might find  [this](https://docs.nvidia.com/cuda/cuda-installation-guide-linux/index.html#axzz3x7mwvZrG) useful

keuntaeklee, 2019-09-24:
I had the same issue after upgrading something and I went back to older versions:

cmake version 3.5.1
gcc version 6.5.0
nvcc version 9.2

and it compiles without error.

closed 2019-09-24 by keuntaeklee