AutoRally/autorally #68: chassisState not published without control input

jsacks3, 2018-07-03:
When I do not use a joystick and instead attempt to manually set runstopMotionEnabled by publishing to runstop topic with the rqt_publisher or a Python script, I found that chassisState was not being published by the simulator. Instead I get the following warning:

```
WARNING: no messages received and simulated time is active.
Is /clock being published?
```

However, when I enable runstopMotionEnabled from the AutoRally OCS and set the control from the Control tab, I can then do a rostopic echo /chassisState successfully. There are no issues when I use the joystick. It seems that the simulator does not publish chassisState unless it receives control decisions.

rdesc, 2020-05-12:
Even if I try sending control commands from the control tab in the OCS, I am still not getting anything published on the `/chassisState` topic. Any ideas?

rdesc, 2020-05-19:
Never mind figured out! Had to set a delay for launching the controller spawner, since gazebo is slow to come up on my machine.

JasonGibson274, 2020-06-12:
This sounds like gazebo not recognizing the controller interface due to namespacing, it should have been address at this point. Please reopen if the issue is persisting.

closed 2020-06-12 by JasonGibson274