AutoRally/autorally #78: Controller Spawner couldn't find the expected controller_manager ROS interface

1140452, 2019-03-29:
Hello I installed your repo and trying to run the `roslaunch autorally_gazebo autoRallyTrackGazeboSim.launch`  and this warn appears: [WARN] [1553860760.215555, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[controller_spawner-6] process has finished cleanly
log file: /home/daniel/.ros/log/03184c32-521a-11e9-af93-1831bf747ba9/controller_spawner-6*.log
[spawn_platform-5] process has finished cleanly
log file: /home/daniel/.ros/log/03184c32-521a-11e9-af93-1831bf747ba9/spawn_platform-5*.log

Do you know how to fix this, because I'm being able to get anything in any simulation to work. thx.

Chufan1990, 2019-04-04:
@1140452 did you fix it? I have the same issue as you.

1140452, 2019-04-04:
Hello, yes i did managed to solve the problem. As I was searching for solutions and it turned out to be memory problems, as the pc couldn't handle all the processing from the simulation. 
To solve it I deleted the cloud rendering they have in _populated_marietta.world_ and it was done.

jsacks3, 2019-04-04:
I have this issue as well. When I manually launch the spawner as per the launch file after Gazebo has been launched, I am able to properly control the car. This problem seemed to appear out of nowhere, as I had not updated anything on my machine. I also don't know why it can't seem to launch the spawner properly within the launch file.

djhanove, 2019-06-16:
Deleting clouds did not work. Had to launch the controller once gazebo finished opening as well @jsacks3

JasonGibson274, 2019-07-10:
Can anyone who is having this issue give me your version of

Ubuntu
Gazebo

I am unable to reproduce this on my machine, but I have seen it happen on others.

1140452, 2019-07-10:
> Can anyone who is having this issue give me your version of
> 
> Ubuntu
> Gazebo
> 
> I am unable to reproduce this on my machine, but I have seen it happen on others.

In my case Its Ubuntu 16.04 and Gazebo 7

JasonGibson274, 2019-07-10:
Can you confirm that this package is installed?  ros-kinetic-gazebo-ros-control

Also that you are on the latest version of kinetic-devel

along with that if you run a launch file with the contents below after gazebo has launched does it fix the issue?

```
<?xml version="1.0"?>
<launch>
  <arg name="cmd_timeout" default="0.5"/>
  <arg name="namespace" default="autorally_platform"/>

  <!-- Load the joint controllers. One of these publishes the joint states to joint_states. -->
  <node name="controller_spawner" pkg="controller_manager" type="spawner"
        args="$(find autorally_gazebo)/config/autoRallyPlatformJointCtrlrParams.yaml"/>

  <!-- Control the steering, axle, and shock absorber joints. -->
  <node name="autorally_controller" pkg="autorally_gazebo"
        type="autorally_controller.py" output="screen">
    <param name="cmd_timeout" value="$(arg cmd_timeout)"/>
    <rosparam file="$(find autorally_gazebo)/config/autoRallyPlatformCtrlrParams.yaml" command="load"/>
  </node>

  <!--node name="wheelSpeeds" pkg="autorally_gazebo" type="wheelSpeedsGazebo.py" output="screen" /Also missing-->

</launch>
```

1140452, 2019-07-11:
I have that package installed and that error does appear when trying to run your launch file. However it stops appearing when I make it to launch the world and the car.

JasonGibson274, 2019-07-11:
So if the world is launched and up before the controllers are spawned it works just fine?

JasonGibson274, 2020-06-12:
Try running with verbose enabled in the track launch file

keuntaeklee, 2020-10-10:
Installing these ros packages solved the same issue:

sudo apt-get install ros-kinetic-ros-control ros-kinetic-joint-state-controller ros-kinetic-effort-controllers ros-kinetic-position-controllers ros-kinetic-velocity-controllers ros-kinetic-ros-controllers ros-kinetic-gazebo-ros ros-kinetic-gazebo-ros-control

JasonGibson274, 2021-02-04:
Closing this issue for now. Please reopen if the problem persists

closed 2021-02-04 by JasonGibson274