AutoRally/autorally #79: Teensy LC Issues on Ubuntu 18

clarkr28, 2019-05-20:
Hello,

We are trying to setup our AutoRally (platform v1.4) on Ubuntu 18/Melodic, and we are currently running into an issue where the Teensy LC that controls our runstop box cannot communicate with any of our machines running Ubuntu 18.  When it is plugged in, it shows up as 'arRunStop' in the /dev/ folder as it should, but the autorally_core RunStop node does not get any messages from it.  The Arduino IDE is not able to read its messages from the serial monitor, nor can Cutecom.  The Arduino IDE is also not able to upload code to it.  However, it works fine on our Ubuntu 16 machines (RunStop node gets messages from it, Arduino Serial Monitor can read its messages, and Arduino IDE can upload code to it).

Do you have any ideas why we might be running into this issue?

bgoldfai, 2019-05-31:
What type of computer are you plugging the runstop box into? This is similar to an issue we have seen with the Teensy communicating with the Dell OCS laptops we have, but that same Teensy is able to communicate with our Lenovo OCS laptops without issue. All of our machines are still running 16.04/Kinetic. So far we haven't been able to track the issue down or confirm if it still exists in 18.04/Melodic, which it now seems like it is still present in some form.

As best as we've been able to debug the issue, we suspect it has to do with the USB chip on the Teensy, but haven't had time to confirm or dig into more detail yet.

Our runtop boxes that are able to communicate with all of the OCS laptops with all Ubuntu/ROS combinations have an Arduino Uno in them. A Micro would also probably work fine as some of our older compute boxes ran with Micros instead of the Teensy to trigger the cameras and we never experienced communication issues.

kentdan3msu, 2019-06-20:
We ended up attempting to fix this issue by installing Ubuntu 16.04 on both our compute box and our OCS laptop, to no avail. It does appear to be the case that the laptop's USB hardware itself is to blame for this issue, not anything on the OS or software level. We're able to work around this by plugging the RunStop into the laptop through an unpowered USB hub. 

On a related note, we had additional issues with the chassis controller (the Arduino Due), but it turns out those issues were related to us using an older version of the hardware BOM that specified a USB isolator with a lower current capacity. We saw the serial stream occasionally output good data if we reflashed it after it powered on, but eventually the device would begin sending more and more corrupt data, and eventually the device would disappear altogether. We are still occasionally seeing corrupted data in the serial stream with the new isolator, but at a much lower rate than what we were seeing with the old isolator, and the device does not appear to be dropping out as it did before. It might be a good idea to add a CRC for the chassis data, but that's a different issue/feature request altogether.