AutoRally/autorally #88: Build fails on Melodic / Ubuntu 18.04 due to StateEstimator linker error (fails to link Boost libraries)

ChrisTimperley, 2020-03-27:
I was unable to build `autorally` on Melodic / Ubuntu 18.04 due to the following linker error when building the `autorally_core` package.

```
[ 92%] Linking CXX executable /ros_ws/devel/lib/autorally_core/StateEstimator
[ 93%] Built target param_getter
[ 94%] Built target pointgrey_camera_node
[ 95%] Built target PointGreyStereoCameraNodelet
[ 96%] Built target pointgrey_stereo_node
[ 97%] Built target pointgrey_list_cameras
[ 98%] Built target PointGreyCameraNodelet
/usr/bin/ld: cannot find -lBoost::system
/usr/bin/ld: cannot find -lBoost::serialization
/usr/bin/ld: cannot find -lBoost::system
/usr/bin/ld: cannot find -lBoost::filesystem
/usr/bin/ld: cannot find -lBoost::thread
/usr/bin/ld: cannot find -lBoost::date_time
/usr/bin/ld: cannot find -lBoost::regex
/usr/bin/ld: cannot find -lBoost::timer
/usr/bin/ld: cannot find -lBoost::chrono
collect2: error: ld returned 1 exit status
autorally/autorally_core/src/StateEstimator/CMakeFiles/StateEstimator.dir/build.make:140: recipe for target '/ros_ws/devel/lib/autorally_core/StateEstimator' failed
make[2]: *** [/ros_ws/devel/lib/autorally_core/StateEstimator] Error 1
CMakeFiles/Makefile2:7472: recipe for target 'autorally/autorally_core/src/StateEstimator/CMakeFiles/StateEstimator.dir/all' failed
make[1]: *** [autorally/autorally_core/src/StateEstimator/CMakeFiles/StateEstimator.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j16 -l16" failed
```

Boost is installed and otherwise working correctly. It seems, however, that the boost libraries are not explicitly linked against via `target_link_libraries`.

To fix the issue, I modified `autorally_core/src/StateEstimator/CMakeLists.txt` as follows:
```diff
@@ -1,4 +1,5 @@
-find_package(Boost REQUIRED)
+find_package(Boost REQUIRED
+  COMPONENTS chrono date_time filesystem regex serialization system thread timer)
 find_package(Eigen3 REQUIRED)
 find_package (GeographicLib REQUIRED)
 
@@ -12,7 +13,7 @@
   include_directories(${GTSAM_INCLUDE_DIR})
 
   add_executable(StateEstimator StateEstimator.cpp)
-  target_link_libraries(StateEstimator ${catkin_LIBRARIES} ${ROS_LIBRARIES} gtsam ${GeographicLib_LIBRARIES} ${TBB_LIBRARIES} ar_diagnostics)
+  target_link_libraries(StateEstimator ${catkin_LIBRARIES} ${ROS_LIBRARIES} gtsam ${Boost_LIBRARIES} ${GeographicLib_LIBRARIES} ${TBB_LIBRARIES} ar_diagnostics)
   add_dependencies(StateEstimator autorally_msgs_gencpp)
 
   install(TARGETS StateEstimator
```

JasonGibson274, 2020-06-12:
I'll make sure this gets address in cmake changes PR currently in progress.

tanmayd26, 2023-08-28:
I tried this but still getting the error.