AutoRally/autorally #104: Running AutoRally in ROS Noetic

Rahul-K-A, 2025-03-23:
Hi,

I've recently come upon this repository, and I want to try running the repo in Ubuntu 20,04 with ROS Noetic. However, the README file doesn't really specify the steps for running the repo in Ubuntu 20. It would be great if someone could reply with updated instructions for Ubuntu 20 and ROS Noetic.

JasonGibson274, 2025-03-24:
We have not updated the repository for 20.04. I think it might be some relatively small and quick fixes, but it will be on you to complete them. The instructions for prior ROS versions should work, just start with noetic installed and go from there. We have few dependencies so it should build pretty easily unless there are breaking API changes from kinetic to noetic.

Are you looking to run just the software (simulation) or are you trying to run on a hardware platform that is upgraded to 20.04?

Rahul-K-A, 2025-03-24:
Hi,

I'm running it on Ubuntu 20.04 with just the simulation. I did get it to work using the following steps:

1. Use Eigen v3.3.7
2. Use GTSAM v4.0.3
3. Continue with the setup in the README as usual
4. For building the ROS workspace, use `catkin_make -DCMAKE_C_COMPILER=gcc-8 -DCMAKE_CXX_COMPILER=g++-8`
5. Replace all the shebangs in the python scripts from `#!/usr/bin/env python` to `#!/usr/bin/env python3`